travisgoodspeed/loraham #8: New GATEWAY: USB doesn't work (when RTC is defined)

akadamson, 2016-12-31:
It seems when you define the RTC into the new gateway code, that you loose the USB port as a usb->serial.  And the reason is because in the sleepsetup() call there is a USBDevice.detach() call.  That probably doesn't do very good when trying to write output to the USB port for Serial.

sharph, 2017-01-03:
`USBDevice.detach()` is called because the USB interrupt wakes up the m0, and the device can't maintain a USB connection when coming in and out of sleep anyway.

We could detect if serial is connected, and if it is, don't sleep and fall back to busy-waiting for timing, but then there would be no way for USB serial to come back in a timely matter. It might even be possible to enable an interrupt on USB that brings the device out of sleep until the USB is disconnected. For simplicities sake, I'm inclined to recommend we add instructions around the `#define RTC_ENABLED` line about the implications of having it enabled. It is intended for the devices on rooftops that should have long battery life and no USB connection.

@travisgoodspeed Thoughts?

akadamson, 2017-01-17:
I just messed with this on another project also using rtcZero for sleep.  Couple of things that I found.  a) you can so a Wire.end(); (if you use i2c), Serial.end(); closes the serial driver code; then a USBDevice.detach();  Go to sleep.  And then, but this next part may only work under linux.  You'd need to do a USBDevice.attach(); delay(seems most use a 1100 number here); Serial.begin(115200); (the baud isn't important, but it appears that under linux and maybe mac OS, the USB-> serial will recover.  Under windows it will not, the renumeration of the USB interface won't occur to allow reconnection to a serial terminal.  Hope this helps.

travisgoodspeed, 2017-02-11:
RTC is now disabled by default in `config.h`.  It should stay that way until this bug is fixed.