travisgoodspeed/loraham #13: Lack of License

jonjmz, 2017-01-14:
I'm having a hard time finding a license for this project. I'm looking to play around with LoRa and have some transceivers on the way, but I don't feel conferrable pulling your code without knowing your intentions for it.

closed 2017-01-17 by travisgoodspeed

brouhaha, 2017-01-17:
I like it, but unfortunately it doesn't meet the Free Software Definition (section 1) or Debian Free Software Guideline (section 1), and won't be acceptable for inclusion in the Debian or Fedora Linux distributions.

travisgoodspeed, 2017-01-17:
Are there any licenses that Debian and Fedora accept which demand proper gratuities to restaurant staff?  If not, which license would you recommend that is acceptable for package inclusion while still politely mocking this modern world's obsession with having to ask permission for every minor thing?

jonjmz, 2017-01-17:
Ha. Works for me travis. This is off topic for this issue, but I was wondering if you were partial to python as the console software. I wanted to get started with fleshing that out but I prefer Golang.

travisgoodspeed, 2017-01-17:
@jonjmz We should probably keep conversations off the issue tracker.  Would you care to join #loraham on Freenode IRC?

Breaking my own rule for one post: The console is in Python because it needs to be a quick example.  Just like the old gateway, we can and will retire it when a proper implementation is available.  If a decent curses library is available for Golang, that would be an excellent choice for a proper console.

The backbone software is already in Golang, and will be slowly merged to this repo as things are ready for external eyes.  The Finger daemon will appear first, maybe later this week.

brouhaha, 2017-01-17:
I don't know of any suitable licenses that meet your objective, sorry.

There was a similar problem with the JSON license, which said that the software shall be used for good, not evil.

In practice, I'm not sure that the license issue is actually a problem. It's unclear to me whether it makes sense to package this for Debian or Fedora.

travisgoodspeed, 2017-01-17:
Okey dokey.  I'll leave this issue closed and re-open it if it becomes a problem.

commit cb25f4e6 (2017-01-17): Added a license.  It's very generous and respects your freedom.

Close #13.