travisgoodspeed/loraham #14: LoRa Modulation Settings

matt-knight, 2017-01-27:
In its current form, `loraham` uses the default modulation settings defined by the Radiohead driver.  This configuration may suboptimal given the project's implicit goals of long range ad hoc networking.  @travisgoodspeed and I have chatted about assessing the current settings and making a change if prudent.

The LoRa modulation has several proverbial knobs that can be turned to trade between throughput and range.  The key parameters, their available values, and effects are:

- **Spreading Factor** ([6:12]): lower == throughput, higher == range
- **Bandwidth** (125kHz, 250kHz, 500kHz): lower == range, higher == throughput
- **Code Rate** (4/([1:4]+4)): lower == throughput, higher == range, optimal == 3 (see my note below)
- **Low Data Rate mode**: on == range, off == throughput

LoRaHAM's current settings are SF 8, BW 125kHz, CR 4/5, and Low Data Rate mode disabled.  These are balanced settings that result in a medium range.

I'm happy to propose more suitable parameters.  If this is of interest, let me lead by asking if there are any time on air or band usage restrictions for the 433 MHz HAM band you've targeted.  This will help scope the rest of the discussion.

matt-knight, 2017-01-27:
Note on Code Rate: The Code Rate option allows you to choose the strength of your Hamming FEC.  (7,4) and (8,4) can correct a single bit error and thus buy you additional link budget, but (7,4) is the only one you should use in practice.  (8,4) can detect a second error but can still only corrects one, thus doesn't get you any performance gains in practice since uncorrected errors are also be caught by the CRC.  The downside of stronger FEC is more overhead, but given your goals it's likely worth the price.

travisgoodspeed, 2017-01-28:
We'll change the rate, but exactly once and never again.  Please start a branch for settings that reasonably maximize range over throughput and we'll merge on a weekend when the whole Philly network can be swapped over at once.

sharph, 2017-01-28:
The library we are using defines several presets that are easy to use. [Otherwise you are required to set registers manually](http://www.airspayce.com/mikem/arduino/RadioHead/classRH__RF95.html#ab9605810c11c025758ea91b2813666e3).

* **Bw125Cr45Sf128** Bw = 125 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on. Default medium range.
* **Bw500Cr45Sf128** Bw = 500 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on. Fast+short range.
* **Bw31_25Cr48Sf512** Bw = 31.25 kHz, Cr = 4/8, Sf = 512chips/symbol, CRC on. Slow+long range.
* **Bw125Cr48Sf4096** Bw = 125 kHz, Cr = 4/8, Sf = 4096chips/symbol, CRC on. Slow+long range.

My thought is that we want to turn the dials as far as we can in favor of longer ranges as long as we're not running into collisions.

matt-knight, 2017-01-30:
If you could please provide the following information I'll chew on the numbers and suggest a configuration.
- Bandwidth limitations as stipulated by 433 MHz HAM rules
- Time on air/duty cycle requirements as stipulated by 433 MHz HAM rules
- Smallest MTU (bytes per MAC frame) you can tolerate, or want to design around

I think that's all I'll need.  I will post again if anything else comes up.

voteblake, 2017-01-31:
Some initial research on @matt-knight's questions. Part 97 governs here in the US of A.

§[97.305](http://www.ecfr.gov/cgi-bin/text-idx?c=ecfr&SID=336ab7469b61ecbfa15086dbf1bf2c59&rgn=div5&view=text&node=47:5.0.1.1.6&idno=47#se47.5.97_1305)(c) states §[97.307](http://www.ecfr.gov/cgi-bin/text-idx?c=ecfr&SID=336ab7469b61ecbfa15086dbf1bf2c59&rgn=div5&view=text&node=47:5.0.1.1.6&idno=47#se47.5.97_1307)(f) paragraphs (6) and (8) apply to the whole 70cm band.

§97.307(f)(6) reads to me that while using a "specified digital code" (in this case ASCII) the "symbol rate must not exceed 56 kilobauds" and "authorized bandwidth is 100 kHz."

§97.307(f)(8) might give us more flexibility as as it allows a whole set of emissions designators defined as follows:
> (H)aving designators with A, B, C, D, E, F, G, H, J or R as the first symbol; 1, 2, 7, 9 or X as the second symbol; and D or W as the third symbol

I believe the three character designator is the legacy form and I'm having trouble putting together exactly what those would mean. Hopefully someone else can lend an eye to the above regulations and also weigh in on the emissions designators.

rrahmani, 2017-02-10:
Given the overall congestion on the 70cm band we should probably err on the side of being good neighbors and keep the transmissions as narrow as feasible

ObjectToolworks, 2017-02-19:
Chirp modulation is a type of spread spectrum, so in that case, there is no bandwidth limit. The narrowband mode limits do not apply. In fact, you probably want about 20 MHz with Direct Sequence. You can also combine Frequency Hopping with that. Since LoRa is chirp, BW limits are on the chip, not in the regulations. The only limit is interference to others, so long chirps are probably problematic.

sharph, 2017-02-20:
I've opened PR #16 with all the dials turned all the way towards range (with the exception of bandwidth, for which I think there is still discussion to be had, and code rate, for which @matt-knight suggested (7,4).) Nothing set in stone; I just wanted to generate some discussion and to have some code we can easily plug numbers into.

I think frequency hopping is a cool idea, but I'm worried the implementation will be tedious and not yield very much benefit. Frequency hopping with these radios requires the radio to interrupt the MCU every so often to ask it for a new frequency to tune to. I think that the library we're using will get in our way. We should consider it if there is a simple and most importantly stable way to implement.

[LoRa Modem Designer's Guide](http://www.semtech.com/images/datasheet/LoraDesignGuide_STD.pdf)

According to the guide, the chip rate is equal to the bandwidth (125kHz = 125kchips/sec.) Narrower bandwidth = more range, lower throughput/longer time on air. If we're going by the examples and equation given in the guide, it appears to be 3dB for every halving/doubling of bandwidth. Depending on if we want to exist in a traditional data channel or if we want to be spread spectrum we might want to push BW to be narrower or wider. If we go narrower, we might want to adjust our spreading factor to account for the time spent on air.

Maybe let's go as narrow as we can, though. Adjusting bandwidth to double the transmission time appears to be 0.5dB better than adjusting spreading factor to double transmission time.

@matt-knight Does this seem right and do you have further suggestions? I don't believe we have any time-on-air restrictions; at least not at the time scale where meshing is possible.

@voteblake @rrahmani @ObjectToolworks, & others who might be more part 97 minded: What should we do? Go narrow? 62.5kHz (the setting under 125kHz?) 7.8kHz?

ObjectToolworks, 2017-02-20:
One thing to consider is the tx power (per Hz) will decrease with the increased bandwidth. So if you really want the spitball to go far, you'd want a nice tight wad. 20 mW over 62.5 is going to go farther than 20 mW over 500 kHz. Setting aside any bit-error comparisons.

But, you also want your 512 bits (64 bytes) to fly fast, so @ 1 sec at 10.4 BW SF=6, .1 sec at 62.5 BW SF=6. 62.5 is looking better than 125 kHz as far as optimum use of the limited power is concerned.

Once you go to the higher SF, the bit rate goes way down, and less nodes will fit. I'm trying to offer some reason to change from the defaults, but I've only tested across the house, where even power spread over 500 kHz works swell. 62.5 at 4.7 kbps with SF=6 seems an optimum power and bit rate for the task.

rrahmani, 2017-02-21:
Giving the SS designator, our operation is covered under part 97.311, and that gives us a lot of leeway as long as we don't interfere with non SS operations. The only aspect I'm not entirely sure of is what the IDing requirement is... previously it was explicitly stated in 97.119, but it was revised out.

matt-knight, 2017-02-22:
@sharph: Thanks for keeping the ball rolling.  Your assessment looks spot on, though I'm confused by the statement `we want to exist in a traditional data channel or if we want to be spread spectrum`.  Do we have input, or is LoRa spread spectrum by definition?

@voteblake and @rrahmani: Thanks for providing great research and insight into the regulations.  Huge help.

**Frequency Hopping**
From conversations with @travisgoodspeed about the goals of the project, I believe frequency hopping is out of scope.  Jump in and correct me if I'm wrong.

**Bandwidth vs. Time on Air**
Going narrow will improve range but will have a serious detrimental effect on time on air, which in turn will reduce maximum throughput and affect the longevity of battery powered nodes.  Additionally if channel congestion within the LoRaHAM network is of concern, a narrowband modulation will exacerbate it.

The attached plot illustrates the tradeoffs between bandwidth and time on air.  These values are for a hypothetical 53-byte length packet (taken from the README: `KC3BVL KK4VCZ Hey Jim, it's Travis.  Care for a beer?`)  Each line represents a spreading factor, code rate is 4/7.

[image]

At 7800 Hz BW, the fastest/lowest sensitivity modulation of SF 7 has a TOA of approximately 2.13 seconds, and that time goes up as SF (and thus sensitivity do)... so it's slow.  Maybe that's OK though; I don't have a sense of how chatty LoRaHAM is now or how busy you expect it to become.  For reference: 144 dB and .242s TOA under the current modulation (BW125000/SF8/CR45), 153.3 dB and 2.13s TOA with the aforementioned hypothetical scheme (BW7800/SF7/CR47).

If Time on Air is less of a consideration, then perhaps a narrower bandwidth and low spreading factor is the way to go: for instance, BW10400/SF8/CR47.  If congestion or battery life is a concern then we can relax the bandwidth and find a compromise.  It's probably a good idea to turn on low data rate mode, although I'm struggling to identify to what extent it affects the link budget.

sharph, 2017-02-22:
@matt-knight I'm a bit unsure of whether we have input myself. LoRa is clearly chirp-based spread spectrum, but since you can configure it at such low bandwidths it starts to look like a digital modulation that could occupy the same amount of spectrum of modes the FCC would classify as "data" in Part 97.

ObjectToolworks, 2017-02-22:
@matt-knight Re: Do we have input, or is LoRa spread spectrum by definition?

The current configuration is for LoRa. To get the device to do OOK/FSK would require many more register configurations (page 87 of RFM95_96_97_98W.pdf, and defeat the benefits of the chirp modulation. Which is to improve range for low power devices. I think the longer chirps might be interesting, although 4096 seems a bit much. Another issue, but in the US wouldn't 446 MHz be more desirable? That is, most mobile antennas are cut for 446 +/-.

As for ID, in the USA you can always ID in the mode you are using. Morse code is deprecated. Not sure about the rest of the planet. But you can always go above and beyond the regulations. ID'ing on the channel is not a good idea. An Internet page would be cool.

ObjectToolworks, 2017-02-22:
Part 97 SS designation (97.3(c)(8) SS. Spread spectrum emissions using bandwidth-expansion modulation emissions having designators with ```A, C, D, F, G, H, J or R as the first symbol; X as the second symbol; X as the third symbol.```

The best description I can find is an FCC compliance test which used the following for LoRa:
```
X - Cases not otherwise covered (Chirp Spread Spectrum)
1 - A single channel containing quantized or digital information without the use of a modulating sub-carrier, excluding  time-division multiplex
D - Data transmission, telemetry, telecommand
```
Preceding this designation is the occupied bandwidth, which is variable. For example, 125 kHz has a 99% bandwidth of 126.7 kHz, so ```127KX1D``` is the designation.

Since the FCC only allows very specific designators on the first symbol in Part 97. We would need to identify the modulation as something better than X. We know it is a chirp spread spectrum modulation, so it is a form of Swept FM. We could call it: ```127KFXX```.

This is another case where technology has passed Amateur Radio Regulations. Maybe the ARRL can submit a change to specify X as an allowable designator. I doubt they care though, since it is way outside their primary interest in legacy modes.

chrismerck, 2017-03-04:
@matt-knight  Regarding bandwidth, keep in mind many users have low-cost crystals (HopeRF modules) which may fail to communicate at minimum bandwidth due to frequency error. I'm unsure of the distribution of this error, but 125kHz+ should be safe considering these modules are intended to at least work for LoRaWAN.

Also, in light of the 70cm band plan, it would be most appropriate to operate LoRaHAM between 420 and 430MHz, which is for experimental and wideband modes like television. Ideally 426MHz.

I do understand that 434MHz is convenient as there are many antennas designed to operate here, but at this frequency we may experience interference in both senses with remote controls under FCC 15.241, and amateur repeater interlinks.

http://www.arrl.org/band-plan

dc2mw, 2017-03-27:
I did some tests with low-cost modules using crystals last year.
[image]

Datasheet tells us that a bandwidth of 62 kHz would be some "magical border" that mandates use of TCXOs. Problems occured mainly in combination with high spreading factors. These very long on-air-times (e. g. 20 seconds for some bytes) would not be useful anyway. 
You can always go two or three steps back (SF or BW or both) from that non-working configuration.

Generally narrow (=less than 125k) bandwidth is possible with crystals.
Further investigation would be needed to get an idea how much margin for extreme temperatures is left.

Reducing BW from 125k to 30k gives about +6dB sensitivity as the noise will drop inside the channel. Rule of thumb: 6 dB increase in link budget gives twice the range.
We are experimenters, aren't we ;-)