travisgoodspeed/loraham #19: Is use of RadioHead mandatory?

rogerclarkmelbourne, 2017-11-16:
Hi

This is not really an issue, but I find the use of RadioHead a bit of overkill for such a simple protocol, and I'm also concerned that potentially RadioHead adds additional data to the start and end of the transmission.

I'm using the much more lightweight LoRa library from Sandeep Mistry

https://github.com/sandeepmistry/arduino-LoRa and it works fine.

Also RadioHead does not AFIK have an official GitHub repo, there is only a mirror that someone unofficially maintains.

skytee, 2017-11-28:
# RadioHead vs LoRa library from Sandeep Mistry 

## Maintenance

The Radiohead library is maintained, releases are provided here: [http://www.airspayce.com/mikem/arduino/RadioHead/](http://www.airspayce.com/mikem/arduino/RadioHead/index.html).

### GitHub vs. Archive
Several people have cloned the Radiohead library on GitHub from a downloaded zip file. Same thing with a GitHub repo, just pull requests are done the old school way: send the maintainer an email. 
No difference if your pull request goes unanswered or if your email goes unanswered.
(Given the maintainer refers to Eric S Raymond on "How to ask questions the smart way" he must have been around the block of open source software development.)

## Transceiver Hardware support
### LoRa library from Sandeep Mistry 
The LoRa library from Sandeep Mistry supports boards based on the Semtech SX1276/77/78/79.

### RadioHead library 
The RadioHead library supports **all those**, plus:

RH_RF22 Works with Hope-RF RF22B and RF23B based transceivers, and compatible chips and modules, including the RFM22B transceiver module such as this bare module: http://www.sparkfun.com/products/10153 and this shield: http://www.sparkfun.com/products/11018 and this board: http://www.anarduino.com/miniwireless and RF23BP modules such as: http://www.anarduino.com/details.jsp?pid=130 Supports GFSK, FSK and OOK. Access to other chip features such as on-chip temperature measurement, analog-digital converter, transmitter power control etc is also provided.
RH_RF24 Works with Silicon Labs Si4460/4461/4463/4464 family of transceivers chip, and the equivalent HopeRF RF24/26/27 family of chips and the HopeRF RFM24W/26W/27W modules. Supports GFSK, FSK and OOK. Access to other chip features such as on-chip temperature measurement, analog-digital converter, transmitter power control etc is also provided.
RH_RF69 Works with Hope-RF RF69B based radio modules, such as the RFM69 module, (as used on the excellent Moteino and Moteino-USB boards from LowPowerLab http://lowpowerlab.com/moteino/ ) and compatible chips and modules such as RFM69W, RFM69HW, RFM69CW, RFM69HCW (Semtech SX1231, SX1231H). Also works with Anarduino MiniWireless -CW and -HW boards http://www.anarduino.com/miniwireless/ including the marvellous high powered MinWireless-HW (with 20dBm output for excellent range). Supports GFSK, FSK.
RH_NRF24 Works with Nordic nRF24 based 2.4GHz radio modules, such as nRF24L01 and others. Also works with Hope-RF RFM73 and compatible devices (such as BK2423). nRF24L01 and RFM73 can interoperate with each other.
RH_NRF905 Works with Nordic nRF905 based 433/868/915 MHz radio modules.
RH_NRF51 Works with Nordic nRF51 compatible 2.4 GHz SoC/devices such as the nRF51822. Also works with Sparkfun nRF52832 breakout board, with Arduino 1.6.13 and Sparkfun nRF52 boards manager 0.2.3
RH_MRF89 Works with Microchip MRF89XA and compatible transceivers. and modules such as MRF89XAM9A.
RH_CC110 Works with Texas Instruments CC110L transceivers and compatible modules such as Anaren AIR BoosterPack 430BOOST-CC110L
RH_E32 Works with EBYTE E32-TTL-1W serial radio transceivers (and possibly other transceivers in the same family)
RH_ASK Works with a range of inexpensive ASK (amplitude shift keying) RF transceivers such as RX-B1 (also known as ST-RX04-ASK) receiver; TX-C1 transmitter and DR3100 transceiver; FS1000A/XY-MK-5V transceiver; HopeRF RFM83C / RFM85. Supports ASK (OOK).

## MCU Hardware support
It is up to the developer what MCU is used and what tool chain for it is used. The code here can be compiled by the Arduino IDE for a plethora of MCUs.

## Code size comparison
Radiohead RFM95 driver: 470 lines of code.
LoRa library from Sandeep Mistry: 523 lines of code.

## Compile size comparison

Minimal example using Radiohead library (compiled for AVR 32U4): 
```
Sketch uses 10760 bytes (37%) of program storage space. Maximum is 28672 bytes.
Global variables use 573 bytes of dynamic memory.
```
Minimal example using LoRa library from Sandeep Mistry (compiled for AVR 32U4)::
```
Sketch uses 5816 bytes (20%) of program storage space. Maximum is 28672 bytes.
Global variables use 214 bytes of dynamic memory.
```

If you were to only support Semtech SX1276/77/78/79 based boards and are strapped for resources, you might care for just using the Radiohead RH95 driver or outright use the LoRa library from Sandeep Mistry .
If you care for support of more transceivers and more features than just LoRaHAM, the Radiohead library looks good.

## Header comparison - Crypto support

The LoRa library from Sandeep Mistry is bound to the same transmitter packet size limitations as the Radiohead library. The LoRa library from Sandeep Mistry supports implicit and explicit headers, while the 
Radiohead library always puts four bytes into the payload to support encrypted mode, too. In effect you lose four bytes payload for a feature you cannot use on HAM bands, but you can use them on the 915MHz ISM bands that some of us use.

## Karma vs. Cost

Supporting Adafruit, SparkFun, Evil Mad Scientist etc. is good karma. If we're talking cheap and abundant, Ali Express is your friend.

## Code Examples
### Example using LoRa library from Sandeep Minstry
```
/* Example using LoRa library from Sandeep Minstry */
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  if (!LoRa.begin(915E6)) {
    while (1);
  }
}

void loop() {
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();
  counter++;
  delay(5000);
}
```
### Example using Radiohead library
```
/* Example using Radiohead library */
#include <SPI.h>
#include <RH_RF95.h>

RH_RF95 rf95;

int counter = 0;

void setup() {
  if (!rf95.init()) {
    while (1);
  }  
}

void loop()
{
  char data[20];
  snprintf(data, 20, "hello %2d ");
  rf95.send(data, sizeof(data));
  rf95.waitPacketSent();
  counter++;
  delay(5000);
}
```

closed 2017-11-28 by rogerclarkmelbourne

voteblake, 2017-11-28:
Hey @skytee, just wanted to say I appreciated this thorough write up!

ReanimationXP, 2018-05-05:
I appreciate it as well. Question though:

For the simple send and receive examples, is it possible to talk cross-Library? Or is there something library-specific going on? It was my understanding LoRa was a standard that should "just work" for simple scenarios without a channel and such, but that doesn't seem to be the case cross-library.

I'm currently trying to send FROM a "HelTec" ESP32 LoRa device with SX1278 radio using the LoRa library by Sandeep Minstry, TO a Moteino Mega w/ RFM95 LoRa radio and the proper RadioHead driver.

Both libraries appear to be doing their thing, but I'm not receiving anything on the HelTec (Sandeep) receiver (server) side. Setting up another Moteino to act as a receiver/server but instead using the Radiohead library demonstrates that said transmitter is in fact transmitting, and I can respond bidirectionally so long as I'm using the same library. I'm on a bit of a time crunch. Anyone know if this is possible or what I'm missing?

ReanimationXP, 2018-05-05:
Sigh, nevermind.. I wasn't the purchaser of the HelTecs and assumed the person who was was abiding by US law. Evidently not. I'll start by getting the right chip before complaining.. lol

[image]

ReanimationXP, 2018-05-07:
Turns out I still had the same issue even after buying the correct radios. Has anyone ever interfaced these two libraries?

PA2PIM, 2018-05-07:
Hi,
I’m assuming you looked at identical frequencies, coding rate, CRC, implicit/explicit header etc, etc. There is one more, not often mentioned, parameter which is different in the default settings between the two libraries and that is the syncword.
The Sandeep Mistry library uses the syncword value of 0x34 as a default. That is the default for a LoRaWAN network.
The Radiohead library as is does not have a function to change the sync word in Register 0x39 so it leaves it at the default of the Semtech Radio chip  of 0x12. i.e. a number for a None LoRaWAN network.
The syncword acts a kind like the network code in GSM, TeTra and LTE network, and it has to be identical for devices to talk to each other.
Try using `LoRa.setSyncWord(0x12);` in your Sandeep mistry device and see if it wants to play ball with the RadioHead device.

Another area where it might go wrong is that the RH library reserves the first 4 bytes of payload for From, To, Flags and sequence number. I believe, but am not sure, that they are already there/reserved if you only use the RH_RF95 driver. I'm not familiar enough with the Sandeep library to tell you how that handles addressing etc.
Pim, PA2PIM

sergionaf, 2018-05-26:
I´m trying to do similar thing.
Change data between a HELTEC LORA 32 using the LoRa library by Sandeep Minstry and a  DRAGINO Lora Gateway using RadioHead library. Until now the devices doesn't interchange data.

Cloolalang, 2020-06-19:
And now there is Radiolib to
https://github.com/jgromes/RadioLib